steadramon/ESPGeiger #1: GPIO configuration from the web UI interface

PetrJakes, 2023-06-17:
It would be great if there is a possibility to set up the GPIO pin(s) for input impulses and for output impulses (testing mode) in teh web UI interface.

steadramon, 2023-06-19:
Hi there, this is something I have considered, especially since building is now automatic through github.

PetrJakes, 2023-06-19:
Tasmota firmware has it for long time and I found it very useful.

krotoniec, 2023-09-26:
Definitely GPIO config would be something very useful.

realatomisator, 2023-10-01:
Thank you very much for this great project. Unfortunately I have Problems with an 38 PIN ESP32 and a cajoe v1.1. 
A GPIO config in the Setup would be helpful.
Thanks a lot!

steadramon, 2023-10-03:
The latest pushes to main has this ability. I will test further and look to make a release

realatomisator, 2023-10-05:
Thank you very much. With the new 0.39 release it works fine!!!

closed 2023-10-06 by steadramon