steadramon/ESPGeiger #2: GPIO defined as INPUT stays high

PetrJakes, 2023-06-20:
I am trying to setup CAJOE IOT-GM-v1.1 counter.
Pulses from counter are connected to the ESP32 GPIO26.

My setup:
platformio.ini: `default_envs = esp32oled_pulse`
counter.h: `#define GEIGER_RXPIN 26`

After the start, the GPIO stays HIGH (3.3V). Because of that ESP can not detect pulses from detector.

Schematic [here](https://github.com/SensorsIot/Geiger-Counter-RadiationD-v1.1-CAJOE-/blob/master/Sch__Geiger%20Counter%20Kit-v1.1.pdf)

Photos of device
[image]
[image]

How can I configure the code so GPIO defined as INPUT stays LOW?

steadramon, 2023-06-21:
Hmm this is one counter that I haven't had access to during development. Is the (PWM for the) high voltage controlled by the ESP32 or is there a separate 555ic, do you know?

When using an ESP32 build by default the internal PCNT is used, the pin should be marked as input by defining what you have.

I'd advise to look into the esp32_pulse_no_pcnt build in environments.ini, which includes the flag  -DIGNORE_PCNT in build flags - this will then use the standard input pin counting method used on the ESP8266. (You can also set GEIGER_RXPIN within a custom environment setting) 

The pin when marked as an input pin should be staying low..

steadramon, 2023-06-21:
Could you get a good photograph of the PCB above, please? Looking at the images above, I can't be sure on the pin that is used by the pulse circuit on the counter above...

PetrJakes, 2023-06-21:
> The pin when marked as an input pin should be staying low..
Exactly.

Can you setup ESP32 as I mentioned (platformio.ini: default_envs = esp32oled_pulse, counter.h: #define GEIGER_RXPIN 26) and test if GPIO26 is LOW after start on your ESP?

I think this is the easiest way how to check it out.

PWM for the high voltage is controlled by separate 555 (see schematic). Only ONE pin is connected/used on ESP. GPIO26

PetrJakes, 2023-06-21:
> Could you get a good photograph of the PCB above, please? Looking at the images above, I can't be sure on the pin that is used by the pulse circuit on the counter above... and I'd like to see the traces

Good point. I was thinking the same and measured by multi-meter yesterday several times. Only one wire goes out from the Geiger board. On the schematic (and on the PCB) marked as INT.

PetrJakes, 2023-06-21:
> Could you get a good photograph of the PCB above, please?

[image]

steadramon, 2023-06-21:
Thanks for your diligence in checking and confirming already 👍 

And thanks for the suggestion, this is something I haven't noticed but it seems that the PCNT option is making the pin high, I need to look into this some more, but -DIGNORE_PCNT may answer the problem?

PetrJakes, 2023-06-21:
I removed  `-DIGNORE_PCNT` flag in environments.ini.
The results are the same. 

New observation: while ESP is waiting for the WiFi configuration, the GPIO26 is LOW. When ESP connects to WiFi, GPIO26 goes HIGH.

steadramon, 2023-06-21:
Hi, sorry, to clarify - you _want_ DIGNORE_PCNT so that PCNT isn't in use

PetrJakes, 2023-06-21:
I am confused. 
As far as I see `-DIGNORE_PCNT` is default in your code. 
In my original setup I made only following changes:
platformio.ini: `default_envs = esp32oled_pulse`
counter.h: `#define GEIGER_RXPIN 26`

It means to me `-DIGNORE_PCNT` was included than. 

Or is it necessary to set up some other flag to include following?

`
[env:esp32oled_pulse_no_pcnt]extends = base:esp32_oled
build_flags =
  ${base:esp32_oled.build_flags}
  -DGEIGER_TYPE=${geiger_type.pulse}
  -DGEIGER_MODEL='"Pulse"'
  -DIGNORE_PCNT`

PetrJakes, 2023-06-21:
sorry i can not format code properly

steadramon, 2023-06-21:
No problem, my previous comment has the environment you should set up at try `esp32oled_pulse_no_pcnt`

steadramon, 2023-06-21:
I may have just pushed a fix for the PCNT issue - the following I found was of use for this https://www.esp32.com/viewtopic.php?t=6346

It seems the PCNT pulls up the pin, interesting I never noticed this before

PetrJakes, 2023-06-21:
I tried `default_envs = esp32_pulse_no_pcnt` which is included in your code

```
Successfully created esp32 image.
Generating partitions .pio/build/esp32_pulse_no_pcnt/partitions.bin
ESPGeiger/src/Counter/Counter.cpp: In member function 'void Counter::setup_pulse()':
ESPGeiger/src/Counter/Counter.cpp:71:54: error: '_geiger_rxpin' was not declared in this scope
   Log::console(PSTR("Counter: Interrupt RXPIN: %d"), _geiger_rxpin);
                                                      ^~~~~~~~~~~~~
ESPGeiger/src/Counter/Counter.cpp:71:54: note: suggested alternative: 'geiger_model'
   Log::console(PSTR("Counter: Interrupt RXPIN: %d"), _geiger_rxpin);
                                                      ^~~~~~~~~~~~~
                                                      geiger_model
*** [.pio/build/esp32_pulse_no_pcnt/src/src/Counter/Counter.cpp.o] Error 1
Compiling .pio/build/esp32_pulse_no_pcnt/lib80b/AsyncTCP/AsyncTCP.cpp.o
```

steadramon, 2023-06-21:
> I tried `default_envs = esp32_pulse_no_pcnt` which is included in your code (I did not made any other changes)
> 
> ```
> Successfully created esp32 image.
> Generating partitions .pio/build/esp32_pulse_no_pcnt/partitions.bin
> ESPGeiger/src/Counter/Counter.cpp: In member function 'void Counter::setup_pulse()':
> ESPGeiger/src/Counter/Counter.cpp:71:54: error: '_geiger_rxpin' was not declared in this scope
>    Log::console(PSTR("Counter: Interrupt RXPIN: %d"), _geiger_rxpin);
>                                                       ^~~~~~~~~~~~~
> ESPGeiger/src/Counter/Counter.cpp:71:54: note: suggested alternative: 'geiger_model'
>    Log::console(PSTR("Counter: Interrupt RXPIN: %d"), _geiger_rxpin);
>                                                       ^~~~~~~~~~~~~
>                                                       geiger_model
> *** [.pio/build/esp32_pulse_no_pcnt/src/src/Counter/Counter.cpp.o] Error 1
> Compiling .pio/build/esp32_pulse_no_pcnt/lib80b/AsyncTCP/AsyncTCP.cpp.o
> ```

That doesn't make sense as the build passes fine for esp32_pulse_no_pcnt - https://github.com/steadramon/ESPGeiger/actions/runs/5335087892/jobs/9667811952 and does locally as well.

_geiger_rxpin is defined here - https://github.com/steadramon/ESPGeiger/blob/main/ESPGeiger/src/Counter/Counter.h#L83 - this should always be defined.... 

As mentioned - the latest version I pushed should pull the input pin low for you - the envs:esp32oled_pulse should now pull the pin LOW after init.

steadramon, 2023-06-21:
I have a couple of the other CAJOE geiger counters that come without the ESP32

[image]

I know these need a resistor adjustment to clean up the pulse that is sent. Is the pin inline with the buzzer? I wonder if adjusting

https://github.com/steadramon/ESPGeiger/blob/main/ESPGeiger/src/Counter/Counter.cpp#L68

`GPIO_PULLDOWN_ONLY` to `GPIO_FLOATING` would help any?

PetrJakes, 2023-06-21:
In general it follows schematic I have sent already
Schematic [here](https://github.com/SensorsIot/Geiger-Counter-RadiationD-v1.1-CAJOE-/blob/master/Sch__Geiger%20Counter%20Kit-v1.1.pdf)

Pulses to the GPIO are going nearly from the TUBE (just an voltage divider and transistor), which is IMHO insane.
I destroyed board while de-soldering/replacing R19.

An other option is a mod to take the signal from D23/LED1.

PetrJakes, 2023-06-21:
> I have a couple of the other CAJOE geiger counters that come without the ESP32

I think the schematic is the same (or very similar)

steadramon, 2023-06-28:
Closing as confirmed functioning code - pin now low after PCNT