steadramon/ESPGeiger #7: Home-Assistant Autodiscovery stops randomly (Unpredictable)

TGCFabian, 2024-04-12:
Ello there!

I've been using ESPGeiger on a cheap BR-6 Geiger-Counter for the past week, 
But i've noticed an issue with the home-assistant MQTT Stuff

I have it send its data over to my HASS, which works fine, sometimes for multiple days, sometimes only for a few hours
it then just, stops (in HASS) and doesnt report anything, it even goes into "Unavailable"

the notable thing is that when i connect directly to the MQTT server, over the same network, i can see the ESP reporting its data just fine, without any issues

To get it back up and running, without rebooting the ESP, i go to the settings page, and simply press "SAVE"
This triggers a new "Auto-discovery" cycle, and fixes the connection again

I'm unsure how to debug this further, so i hope this helps out alot

Tiny things i tried
```
Stable 5v Input to the ESPBoard (Using its own 3.3v regulator)
Stable 3.3v Input
Different networks (Never know!)
Different usernames (ofc shouldnt matter, but still worth a try)
Reflashing ESP from clear
```

steadramon, 2024-04-12:
Studying the HA Autodiscovery documentation again, I think I've spotted a potential cause/solution. Leave it with me.

steadramon, 2024-04-13:
@TGCFabian - thanks for the comprehensive bug report and steps you've tried. 

[v0.5.4](https://github.com/steadramon/ESPGeiger/releases/tag/v0.5.4) should resolve this issue for you. For information, the ESPGeiger now subscribes to the default HA LWT status topic (`homeassistant/status`) and re-publishes configuration as required.

If you could test and let me know that would be great

TGCFabian, 2024-04-13:
I've installed it on my ESP, and will report if any issues happen in the next 48 hours, 
Thanks for the quick fix!, will report back soon!

TGCFabian, 2024-04-15:
Hi there, here to follow-up

@steadramon You've indeed fixed the issue that was happening, it's been running perfectly fine, and reporting all data without issues

Thank you so much for the quick responses, and the quick-fix!

closed 2024-04-15 by TGCFabian