steadramon/ESPGeiger #8: Problem with the OLED display

georgi-m-iliev, 2024-06-16:
Hello!

First of all I want to say congrats on the amazing work you've done on this project. It is really complete and has awesome features. That is way I really want to use it to power my geiger counter.

The issue I'm experiencing is the display not clearing.
[image]

I am compiling my own version due to different pins used for the I2C interface. I've tested the screen with the Adafruit_SSD1306 and the screen is working as expected.

Would you have any idea what might be the issue and how to resolve it?

steadramon, 2024-06-17:
Hi there,

Can you confirm the alternate pins you're using? Also the board on which this is set up, so that I might be able to replicate?

It does feel like the clear() function isn't updating the screen properly.
This could be related to the clock on the alternate pins - this is something I looked into a while ago, but have since used the assigned I2C pins on the boards.

You could try uncommenting this line and try adjusting clock speed - https://github.com/steadramon/ESPGeiger/blob/main/ESPGeiger/src/OLEDDisplay/OLEDDisplay.h#L62

georgi-m-iliev, 2024-06-18:
Hi, thank you for responding!

I am using the LOLIN D1 Mini (v4) board and the following pin arrangement:
SCL -> 0; SDA -> 2; GEIGER_RXPIN -> 4.

I tried different values for the clock speed, a lot lower and a higher than the default one, but to no avail, the issue persists. It is possible the pins are not suitable for use with I2C, but the weird thing is that the Adafruit library works just fine. As a note, I am using those pins, because for some reason, the only pin that was able to be used for interrupt when pulled HIGH, was pin 4, which is the default SDA pin, on the other hand, for some reason I was never successful in using pin 5 as a digital output - probably some defect on the board itself.

steadramon, 2024-06-22:
Hi I've been looking through this - I cannot replicate.. it's mighty weird as it seems that some of the update functions are only partially running.. running some items within the function but not others - this isn't just limited to "clear"w

For example, I notice that you seem to have the historic graph drawn, however the axes for the graph aren't drawn - in the function, the axes are drawn just before the historic graph. The min/max values are also not on the graph - these are drawn immediately after the graph is drawn.

I'm very confused with this. I've tried several ways of breaking this in the same way - I've possibly spotted a few bugs but nothing which would cause such weird behaviour. 

Have you the ability to test with different pins or another board?

Can you share your full env definition (and base if there) just in case there's anything there I'm assuming/missing?

steadramon, 2024-06-22:
Sorry about this I think I've realised what the issue is - pin 2 is used for the inbuilt LED on the board, after digging out a v4 and soldering it up the same way, I noticed the problem. 

Use a different pin than pin 2.

georgi-m-iliev, 2024-07-08:
That really is it! I hadn't realized that ESPGeiger uses the onboard LED and that results in interference with the I2C communication. Having moved the OLED display to another set of pins has resolved the issue and everything works just fantastic. I am really glad that hardworking and devoted people like you exist! Thank you again for the awesome project!