steadramon/ESPGeiger #9: Just a few requests and remarks.

SargonofAssyria, 2024-06-29:
First nice job!
It would be nice to have the possible switch between 24h and am/pm in the graph.
My Access Point SSID is not readable in menu Info,
It would be nice if switch/display of summertime/wintertime is automated.
It would also be nice if current date/time is included in ESPGeiger-XXXXXX/tele/status
Have you seen the dial presentation of https://github.com/mkgeiger/geiger-counter-iot it would be nice if that could be included.

steadramon, 2024-06-29:
>  It would be nice to have the possible switch between 24h and am/pm in the graph.
Pull requests are welcomed! :) 

>  My Access Point SSID is not readable in menu Info,
I rely on WifiManager to manage this - https://github.com/tzapu/WiFiManager

>  It would be nice if switch/display of summertime/wintertime is automated.
If you set the Timezone within the NTP settings, it should be. Where are you seeing that it isn't?

>  It would also be nice if current date/time is included in ESPGeiger-XXXXXX/tele/status
I can see this is something Tasmota seems to implement, seems strange to me as the broker should be aware of the timestamp that the message entered the queue, I'll look at this.

>  Have you seen the dial presentation of https://github.com/mkgeiger/geiger-counter-iot it would be nice if that could be included.
Yes, the ESPGeiger web interface is fully self contained - the above makes use of Google Charts, which I will not implement. I may consider something along these lines when the warning and alarm values are made adjustable.

steadramon, 2024-06-29:
> It would also be nice if current date/time is included in ESPGeiger-XXXXXX/tele/status

853df59 should resolve

SargonofAssyria, 2024-06-29:
Tnx, for the date. Suddenly the SSID is displaying normal,. so solved.
I set the timezone to Europe/Brussel now, that solves it, but Europe/Amsterdam should be present too.

steadramon, 2024-06-29:
Thanks for the feedback.

At present ESPGeiger is mostly only supporting the core Timezone sources, mostly due to the sizes resitrictions of the ESP8266. The same timezone details "CET-1CEST,M3.5.0,M10.5.0/3" would be applied.

https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

closed 2024-08-09 by steadramon