steadramon/ESPGeiger #10: Feature request - configurable interval for logging to services

Mr-Blinky, 2024-07-16:
It would be very handy to be able to configure the logging interval for external services such as MQTT and radmon.org, and possibly the others.

I would like the ability to send data to MQTT every 10 seconds and some radmon.org users/stations submit readings at intervals other than 1 minute. The minimum submission interval for radmon.org is 30 seconds as any submissions sooner than 30 seconds are dropped by radmon core.

I'm not sure about the other services (HA, Thingspeak and GMC) as I have no working knowledge of them, and a configurable interval may or may not be useful to them.

I also understand that if just one submission timer is used in ESPGeiger then implementing this would be quite a job.

Cheers

(Simomax @ radmon.org)

steadramon, 2024-07-17:
Hi 😸 Thanks for the report.

Configurable submission time of statistics to the online services is something I have on my roadmap. At the same time I am looking at the configuration management and variable storage within ESPGeiger - this might be a good opportunity to also add these options.

For MQTT the main variable will configure how often the "/stats/" topic values are posted, "/tele" MQTT posts will likely have a different configuration option.

One thought regarding different submission times to Radmon and the like would ideally require detection of the submission frequency and then submitting the most appropriate CPM bucket value - 1/5/15 minutes.
This shouldn't be an issue, more a note.
I think upper and lower limits should be implemented on ESPGeiger regardless. I want to be sure that the scope for abuse is limited within the UI etc.

steadramon, 2024-08-09:
Release [v0.6.1](https://github.com/steadramon/ESPGeiger/releases/tag/v0.6.1) has implemented adjustable submission time for MQTT and Radmon.

Mr-Blinky, 2024-08-30:
I have updated to 0.6.1 and this works great. I have tested with radmon.org and MQTT time intervals and can confirm they are working as expected.

There is one small issue with the radmon.org timer (and maybe any others with limits). I can see (and verified) the minimum submission time for radmon.org is 30, but if I enter 5 seconds the submission is 30 seconds but the form field on the config page will display 5 seconds (please see image). I am assuming you are storing the variable as '5' and the hard limit in code is maintaining the 30 second limit. It would be nice to display 30 seconds if a time of less than 30 seconds is entered.

Cheers

[image]

steadramon, 2024-08-31:
I have placed a validator on the text field which will ensure it is within the values. This is partly due to the way the configuration items are handled, as mentioned above, but implemented using the existing configuration handling. I'll put out a release that uses the input limit first, then I do need to sort the configuration handling for things like this.

steadramon, 2024-09-02:
Release [v0.6.2](https://github.com/steadramon/ESPGeiger/releases/tag/v0.6.2) has adjustments for santisation of user input - you may need to resubmit on the config page for this to be effective

Mr-Blinky, 2024-09-07:
I have updated to v0.6.2 and can confirm this is now working as expected. Nice work. Thanks! I guess this issue can now be closed.

closed 2024-09-07 by Mr-Blinky