steadramon/ESPGeiger #13: Cajoe image not working

shyrwall, 2024-08-16:
Just got a cajoe esp32-gc board and the image v0.6.1 is not working in PCNT. However NO_PCNT is working. 

Since I never used espgeiger before I don't know if it worked in releases before 0.6.1

steadramon, 2024-08-17:
Hi there - thanks for the report. I had a similar query several months ago - I thought the issue had been resolved, this could be related to the Pin mode set after the PCNT is initialised.

I don't have a Cajoe ESP32 to test with, but I do have something we can try before relegating the Cajoe build to NO_PCNT.

Can you try the latest esp32_cajoe_iotgm_test_pcnt build from the pipeline, please?

https://github.com/steadramon/ESPGeiger/actions/runs/10430810811/artifacts/1822671082

Thanks!

shyrwall, 2024-08-17:
Works.

Did the upgrade with firmware.bin via the webinterface. Version after came up as devel/0558524

Thanks

steadramon, 2024-08-17:
That's great news - thank you for testing - I'll get this out in an official release soon - but this is exactly the same as 0.6.1 except with this fix.

If possible, would you be willing to take a photo or two of the Cajoe device running the ESPGeiger firmware? I'd really like to use these for the documentation page

You should be able to upload them here if you're able,

Thanks!

Paul

shyrwall, 2024-08-17:
[image]

steadramon, 2024-09-02:
Release [v0.6.2](https://github.com/steadramon/ESPGeiger/releases/tag/v0.6.2) has these specific fixes for the Cajoe ESP32-GC image

closed 2024-09-02 by steadramon