steadramon/ESPGeiger #16: Please Help -->Serial Connection with ESP D1 Mini and Cajoe Geiger Counter v.1.1

raindancer2204, 2024-09-13:
Hello,

after the display now works with my D1 Mini (I use the "Default Pulse WEB Image" I tried the serial
Connection to my Cajooe Geiger Counter v1.1. to build.

The Geiger counter clicks (sound) and also seems to detect radiation.

The Geiger RX pin was tested with both D7 and the RX on the D1 Mainbaord. Unfortunately I can't receive a signal.

What should i put here?
[image]

Do I have to set any jumpers or switches on the Cajoe V1.1?

How does the test work? I bridged tx and rx on the D1 Mini Board. Unfortunately that didn't show anything either.

It would be nice if I could get an answer. I'm so close to the finish line :)

steadramon, 2024-09-13:
Connecting the Cajoe's "VIN" pin to D7 (GPIO13) on the D1 Mini is correct as per your "RX Pin" definition as shown.

Do make sure that your D1 mini is also connected to the common ground from the Cajoe counter. This should be enough to get it working.

`VIN` -> `D7`
`GND` -> `GND`

There shouldn't be a need to set any jumpers etc on the Cajoe counter.

---

The ESPGeiger Test Pulse pin is only available on the Test Pulse build of the firmware - the standard Pulse firmware does not have a test pin output.

[image]

If you were to install the Test Pulse firmware this will generate a test signal on `D6` (`GPIO12`) (**GeigerTX**), you can then connect `D6` (`GPIO12`) -> `D7` (`GPIO13`) (**GeigerRX**) to show the test signal.

raindancer2204, 2024-09-14:
I checked it with this firmware and now it works. 

Could you please mention this in 1 to 2 sentences in the "**Serial counter**" section in the RL.

For others it would be important that the "test-impluse" only works with this firmware. In addition, this can also affect the test signal. With a short example.

I've spent some time here now and finally opened the issue. We could save others time that and move the project forward.

Would it be of any use to you if I provided the PDF with the information? You could then bring it to the main page.

So i can support with that.

I would modify it a little.

Thanks in advance for the quick help.

steadramon, 2024-09-14:
Sorry I don't understand your request. The Test Pulse pin is only included with the Test Pulse build. The builds are separate so as to not affect a real/live counter.

The Cajoe v1.1 is a Pulse based counter, for the default pins as set, the information in section 2 of the readme looks fine. I'm not sure what your reference to the Serial counter section is in regards to?

More comprehensive documentation is also available at https://docs.espgeiger.com/ - however this isn't 100% complete as of yet.

Basti149, 2024-09-14:
Hello,
I also have a CAJOE Radiation Sensor which I use with a Wemos Mini (ESP8266). But unfortunately it does not work like the raindancer2204
I have connected the pulse output of the Geiger counter to the D7 of the Wemos (correctly). But now the software does not receive any data.
The Geiger counter makes noises, so it works, but the ESP doesn't receive anything.
Do I have to enter any more data in ESPGeiger? Or are the standard settings OK?

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

Basti149, 2024-09-14:
How should the jumpers/switches be positioned on the board?

steadramon, 2024-09-14:
No further settings should be required as the cajoe is a pulse counter.

Can you send a photograph of your setup?

Basti149, 2024-09-15:
@steadramon 
yes too

[image]
[image]
[image]
[image]

raindancer2204, 2024-09-15:
Great i think that is the case from thingiverse right? [https://www.thingiverse.com/thing:5982201](url) . I will do the same way. Should also be in the README

steadramon, 2024-09-15:
Maybe there's some confusion /lost in rtranslation happening here as from image #3 with case assembled seems to show that the graph is displaying a current CPM value - this would indicate this is working.

raindancer2204, 2024-09-15:
Do you happen to know how to turn off the beep on the Cajoe? The plan would be to solve this with a switch. So you can switch it on and off.

steadramon, 2024-09-15:
J1 on the Cajoe board controls the beep. Yes I have used different tubes with the Cajoe, however not one with alpha detection.

To confirm - the standard Pulse firmware is working for you? What did you change?

raindancer2204, 2024-09-15:
On the subject of tubes, I read with the capital A in the reviews that the "SBM20" only shows beta and gamma radiation. That's why I asked if you've ever used them.

Yes, it really works great for me with the Puls firmware, thank you very much!

steadramon, 2024-09-15:
I have a number of SBM20 and STS5 tubes in use with my hardware Geiger counter (ESPGeiger HW) - as you say these are only beta and gamma radiation - great for background monitoring

As far as I know there isn't a tube which detects alpha and will fit into the Cajoe without modification - even with modification I don't know if the power output of the Cajoe would be enough - the LND712 requires ~500V operating voltage and a majority the pancake detectors require circa 700V.

Some of this might be better in the discussion area or over at the Radmon forums.

Basti149, 2024-09-15:
@steadramon @raindancer2204 
Thanks, I'll have to try the tips listed.
Thank you both.

steadramon, 2024-09-15:
@Basti149 - your image 3 show a CPM value - this would seem to be working.

You might want to connect to the web interface for the ESPGeiger? If you made a note of the ESPGeiger ID - visit http://ESPGeiger-XXXXXX.local - or look in your router for the IP address given to ESPGeiger by DHCP.

Basti149, 2024-09-15:
@steadramon 
Thanks, I wrote down the IP address from my router. I'll try it again tomorrow in detail or flash the ESP again and do everything step by step. Maybe it will work then. But the jumpers/switches are correct as I set them and the output from the Geiger counter on d7 is also correct.

tinovator, 2025-01-02:
Hi @steadramon! I tried to implement this project today but I face the same issue as the OP. 

I am using ESP32 with CAJOE v1.1. 
I tested with my own code to count how many ticks are there in a minute and it checks out ok regarding backgroudn radiation in my area. 

No matter what pin I put in setting it never picks up any ticks from Geiger. 

I remember that one time while I was programming simmilar thing happened to me when I declared geiger input as pinMode(geigerPin, INPUT_PULLUP). If it was declared just as pinMode(geigerPin, INPUT) everything would work fine. 

What would be possible solutions to overcome this?

steadramon, 2025-01-03:
Hi @tinovator  - 

The original poster was using an ESP8266 and the Serial firmware. I have copied your post to #21

closed 2025-01-03 by steadramon

raindancer2204, 2025-05-29:
I have a addional question i want to update my ESP Geiger via httpd local url. My Firmware is really old Build date
Sep 2 2024 18:47:07.... which should i use for update? I dont want to use the webinstaller.