steadramon/ESPGeiger #21: ESP32 - Cajoe v1.1

steadramon, 2025-01-03:
Hi @steadramon! I tried to implement this project today but I face the same issue as the OP. 

I am using ESP32 with CAJOE v1.1. 
I tested with my own code to count how many ticks are there in a minute and it checks out ok regarding backgroudn radiation in my area. 

No matter what pin I put in setting it never picks up any ticks from Geiger. 

I remember that one time while I was programming simmilar thing happened to me when I declared geiger input as pinMode(geigerPin, INPUT_PULLUP). If it was declared just as pinMode(geigerPin, INPUT) everything would work fine. 

What would be possible solutions to overcome this?

_Originally posted by @tinovator in https://github.com/steadramon/ESPGeiger/issues/16#issuecomment-2568269207_

steadramon, 2025-01-03:
Hello - 

Can you confirm the version of the firmware you are using - the commentor you posted on was using the GC10 serial firmware on ESP8266 - the Cajoe is a Pulse based Geiger, therefore the Pulse based firmware is required.

By default the pulse input should be connected to `GPIO13`

You could try both the **Pulse** and **Pulse - No PCNT** builds for the ESP32.

tinovator, 2025-01-04:
It looks like this version is running fine and I assume it si because of different pinMode as I have tested before. Thank you! 

There are two additional issues I noticed:

1. When adding or editing credentials for Radmon/GMC/Thingspeak upon saving, page leads and I get ERR_CONNECTION_RESET This site can’t be reached error in my browser. It seems like the system accepts the date or changes made and everythinng works but this blank screen always happens. I've tried Chrome, Safari and Firefox on MacOS. 

2. Can you please just check what is happening with installer because I was not able to install Pulse - No PCNT with OLED screen, installation passes only if I do not select OLED screen installed. I have spare OLED which I would like to use. 

[image]

Additional question: 
1. Is there any significant difference between Pulse and Pulse - No PCNT version?

steadramon, 2025-01-05:
> When adding or editing credentials for Radmon/GMC/Thingspeak upon saving, page leads and I get ERR_CONNECTION_RESET This site can’t be reached error in my browser. It seems like the system accepts [...]

This is the first I've heard reports of this and I cannot replicate on my end - I too use MacOS primarily. I can try a few other ESP32 variants but my go-to's seem fine - does this only happen after initial flash or is it persistent?

> Can you please just check what is happening with installer ...

This should be fixed.

> Is there any significant difference between Pulse and Pulse - No PCNT version?

The Pulse version on ESP32 uses the ESP32 [hardware pulse counter](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/pcnt.html) - the "No PCNT" version uses a basic Pulse detection mechanism, the same as is used in the ESP8266 build.

steadramon, 2025-01-07:
I've just re-published the ESP32 OLED No PCNT - this looked to be the wrong firmware - please try again.

`esp32oled_pulse` looks fine, though

If this doesn't help -

- Do you get this while installing?
- Do you get this with no other devices (OLED, Geiger counter) connected to the ESP32?
- Try installing the "firmware_merged" versions from the latest release:

`esp32oled_pulse` - https://github.com/steadramon/ESPGeiger/releases/download/v0.6.2/esp32oled_pulse-firmware_merged.v0.6.2.bin
`esp32oled_pulse_no_pcnt` - https://github.com/steadramon/ESPGeiger/releases/download/v0.6.2/esp32oled_pulse_no_pcnt-firmware_merged.v0.6.2.bin

tinovator, 2025-01-07:
I tried flashing again and now everything works without error. Both Pulse and no PCT version.

What would be the right pins to use for ESP32? 
I tried:
SCL - 22
SDA - 21 
tried also combination of D4 and D5.

My screen is 128x64 0,96" with address 0x3c. Previously in other sketches I used Adafruit_SSD1306.h library and SSD1306_SWITCHCAPVCC to initialize. 

Any ideas?

steadramon, 2025-01-08:
The firmware has the OLED pins hard-coded to:

`OLED_SDA - GPIO4`
`OLED_SCL - GPIO5`

This is something that will be confiurable in a future release

tinovator, 2025-01-08:
Ok it looks like it will not work due to using Adafruit library which works for me on test sketch but is not included in your. 
Do you have any advice on what to try next or can you recommend another OLED that I could purchase that would work in this project?

steadramon, 2025-01-09:
I'm not sure how the SSD1306_SWITCHCAPVCC would affect your display - I essentially plug the OLED into the GND, 3.3v and GPIO4/5 as above.

The screen you describe sounds much like the ones I have tested and used - generic SD1306 screens from Alie* - see https://github.com/steadramon/ESPGeiger/discussions/17 for an example of a larger display with ESP8266

Have you tried the previous comment with both pins (and reversed - just in case!)?

I've honestly yet to come across a "SSD1306" that hasn't worked out of the box but there's certainly a chance of it happening

tinovator, 2025-01-09:
I tested my screen from scratch with different sketches and finally it started working after couple of tries. 

I think it is safe to say you can close this issue now :) 

Thank you for all your help making this project possible!

[image]

closed 2025-01-10 by steadramon