steadramon/ESPGeiger #23: V.7.0 will not run on an ESP32

ChrisLX200, 2025-04-24:
**Describe the bug**
Loading ESPGeiger onto ESP32-WROOM-32 (various versions tested) causes a loop at startup. Failure to enable WiFi.

**Version of ESPGeiger Affected**
Tested several versions from latest V.7.0 back to V.5, all have the same issue

**Geiger Counter in Use**
Cajoe v1.1

**To Reproduce**
Simply load it onto an ESP32 and observe the console output!

Have successfully loaded and run the code on an ESP8266 so it is not my upload methodology.

ChrisLX200, 2025-04-24:
Yes I tried both. Also tried using the Web Installer, and the downloaded .bin file, and tried compiling my own from Platformio. I did not try all old versions, just a 6.x and the 5.0. Struggled a bit with platformio as I'm not familiar with it but got there in the end. Also tried to transfer the source to Arduino but one of the libraries (the particular 1306 display library I believe) I could not find. I was trying to find the bug but I;ve given up on that idea!

ChrisLX200, 2025-04-24:
Tried the 0.7.1 OledPulseMerged bin file. Loaded OK. WiFi loop fixed. Goes into a loop trying to read the flash. I had cleared the flash before loading the file onto the ESP32.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

ChrisLX200, 2025-04-24:
Ignore above error report - I was using ESPTools to flash the bin file and flashed it at offset 0x1000 instead of 0x0...

ChrisLX200, 2025-04-25:
The OLED displays on my ESP32 boards are soldered on, I have absolutely no documentation for these dev boards but I do have some code that works with them. The address/pin allocation (SDA and SCL respectively) is    SSD1306 display(0x3c, 5, 4);

ChrisLX200, 2025-04-25:
Made the changes for SDA/SCL locally and recompiled. OLED now working.

steadramon, 2025-04-30:
I have considered a configuration option for the OLED pins. This will be forthcoming in a future update, with updates to the Preferences storage overall. Thanks!

intro74, 2026-01-21:
Hi, I have the same problem on an ESP32-WROOM-32 with SSD1306: boot loop at startup and “Failure to enable WiFi.”
Did you manage to solve it?